LibreSolar/mppt-2420-rc #1: Capacitor rev. 0.1

BernhardEllmann, 2020-06-13:
Dear Martin, 

both circuit diagrams rev. 0.1 and rev. 0.1.1 specify C1 + C2 as 560 uF
BOM for rev. 0.1.1 lists 560 uF/100V
you commented that footprint of C1 + C2 should be reduced when building the circuit on board version 0.1

do you recommend:
- to reduce 560 uF --> ?
- or to reduce 100V --> ?
- or both 560 uF / 100V --> ? / ?

my use-case is:
- solar panel 21,4 V OC, 17,5 MP 
- battery 36 V

can you please confirm that the controller can handle this? The diagram does not mention that the circuit can actually be used in 2 directions

martinjaeger, 2020-06-14:
Yes, boost operation is possible, as it is a synchronous buck converter.

For the nominal 36V battery (i.e. 42V max.) also 63V or 80V caps should work fine. Slightly lower capacity (like 390uF) is also OK. Most important is that the caps allow high ripple current. If you take one from the same series as specified in the BOM everything should be fine.

BernhardEllmann, 2020-06-14:
fine, thanks a lot!

closed 2020-06-14 by martinjaeger