LibreSolar/mppt-2420-rc #3: BOM: some faulty part numbers

BernhardEllmann, 2020-06-13:
some part numbers in BOM, .csv-version, appear to be faulty:

- line 8 (C15): part GRM188R71H104KA93D has wrong value (0,1 uF instead of 1nF
- line 39 (R12): part RC0603FR-0710KL has wrong value (10K instead of 0R)
- line 43 (R64, R65, R66): part RC0805FR-0710KL has wrong footprint (or wrong data specified in column F)
- line 47 (R8, R10, R14): part RC0603FR-0710KL has wrong value (10k instead of 2k2)
- line 52 (R60): part RC0805FR-0733KL has wrong footprint (or wrong data specified in column F)
- line 55 (R9, R11): part RC0603FR-0710KL has wrong value (10K instead of 68K)

martinjaeger, 2020-06-14:
Thanks for finding these errors. I didn't realize them as I have all those standard parts available anyway and don't order them with each PCB.

- Line 8: Ok, taking out the specific part number, as the manufacturer doesn't matter for this part.
- Line 39: Ok, corrected.
- Line 43: Part numbers were wrong, 0603 is correct.
- Line 47: Ok, corrected.
- Line 52: 0603 footprint is correct, part number was wrong.
- Line 55: Ok, corrected.

Will push the corrections with next update.

BernhardEllmann, 2020-06-14:
Line 8: so the same type can be used as for C6, eventually? (Same footprint, same value)
If yes, the two lines 8 & 9 could be merged 

btw:
C15 got changed from v0.1 to v0.1.1 (from 100 nf to 1 nF)
I guess for biuiding the circuit on board version 0.1, I shall use the updated value?

martinjaeger, 2020-06-14:
Yes, correct, 1nF can be used and the same caps can be used for both.

The snubber capacitor C6 has some influence on voltage ringing and EMC behavior and still has to be fine-tuned in the final design, so I did not merge both of them yet. But at the moment we can use the same caps for both (even though C15 doesn't need that high voltage rating).

closed 2020-06-26 by martinjaeger