LibreSolar/mppt-2420-rc #5: BOM incomplete (MOSFET clips)

BernhardEllmann, 2020-06-13:
BOM appears to be incomplete:
MOSFET clips for heatsink are not included.

martinjaeger, 2020-06-14:
Thanks for these remarks. Those mechanical parts were currently not part of the BOM as KiCad does not have any feature to add parts that are not placed on the board... I will think about a way how to do that without adding them manually to the BOM after export.

The correct Heatsink clip is THFU 2. I will at least add it as a remark. Insulating pads are specified in detail yet, I just took one that I still had. For the fuse you can use Littelfuse 0287025.PXCN.

BernhardEllmann, 2020-06-14:
Why do you recommend THFU 2?
THFU 1
- is suitable for TO 220
- is suitable for SK 573
- has same strenght (60 N)
- is a tiny bit less bulky, according to technical drawing on Fischer Website (10,3 mm instead of 10,5 mm)

Have you tried THFU 1?

martinjaeger, 2020-06-14:
Good question. That one was recommended to me by Fischer Elektronik when I talked to them at Embedded World this year, so I didn't check any other option.

The THFU 2 was quite hard to mount, you needed lots of force to clip it in. Maybe THFU 1 is a bit easier to mount... but didn't try.

BernhardEllmann, 2020-06-14:
ok, I will order both and let you know :)