LibreSolar/mppt-2420-rc #6: replacement for heatsink Fischer SK 573 37,5

BernhardEllmann, 2020-06-13:
Heatsink Fischer Elektronik SK 573 37,5 seems to be hard to get.
is there any alternatives from other manufacturers?

martinjaeger, 2020-06-14:
Ok, that seems true. Only Arrow seem to have it in stock: https://octopart.com/search?q=Fischer+Elektronik+SK+573+37%2C5&currency=EUR&specs=0

I got a sample directly from Fischer Elektronik.

If you don't populate the UEXT connector J7, quite a lot of other heat sinks should be useable if the MOSFETs are screw-mounted to the heat sink. This particular one fits very well into the available space and doesn't require threads in the heat sink as it uses clips. I didn't do any research if other heat sinks fit as nicely as this one. As stated in the Readme, I still have to validate if the heat dissipation inside the housing is enough for the 20A in general.

BernhardEllmann, 2020-06-14:
the longer 50mm version is available at Farnell.
using that one after trimming would be a workaround

martinjaeger, 2020-06-14:
Ah, yes, black is recommended.

BernhardEllmann, 2020-06-14:
"If you don't populate the UEXT connector J7 ..." (cited from above)
What is the minimum setup required for programming the MC when using the v0.1 board
SWD and USART port?
(There is no debug serial port available on v0.1)

martinjaeger, 2020-06-14:
Only the SWD is needed for programming. But an USART is useful for debugging. For the v0.1 board I am using the USART in the UEXT connector, but as there was another USART left, I added a second one to the v0.1.1 board.

BernhardEllmann, 2020-06-14:
This means that you recommend to populate the UEXT port, did I understand that right?

I'm asking, because on the schematic (PDF, Page 4) the keyword "USART" shows up in combination with the keyword SWD: "STM Nucleo SWD and USART" (blue text)
(sorry. since I'm not too familiar with these ports yet, it's a little bit confusing for me)

martinjaeger, 2020-06-14:
The UEXT port is very useful, as you can add lots of extensions to the board, see here: https://www.olimex.com/Products/Modules/

However, if you need the space for a different heat sink and still want to use the USART for debugging (which I highly recommend), you can also solder those two wires to the correct pins at the UEXT connector only.

The text refers to SWD and USART at the same time as both are quite useful during development and most programmers / debug probes (including the ST-Link v2 as included in many ST nucleo boards) also have the USART pins that you can use for printing serial debug information.

BernhardEllmann, 2020-06-14:
sounds good, thank you!