LibreSolar/mppt-2420-rc #7: BOM consolidation: C8, C14, C16, C34 vs. C17

BernhardEllmann, 2020-06-13:
Why use different types for:
- C8, C14, C16: 50V
- C17: 100V

could 100V type be used for all 4 components?

martinjaeger, 2020-06-14:
Thanks for these enhancements.

- C17 etc. (10 nF): I agree this makes sense. Will update this.
- C41 etc. (2.2uF): I forgot to add the part number in the schematic for one of them, so the BOM export puts it in a separate line as their information was not *exactly* the same. Just added the number also for C41, so it will get to the same line for next export.
- D2, D3: Oops, forgot to specify as I just took one of the SS14FL. We could actually consolidate the different types of Schottky diodes better and just use 60V / 2A versions everywhere. Suitable parts: 
Nexperia PMEG6020ER,115 or ON Semi NRVTS260ESFT1G
- J3, J4: Amphenol FCI 54602-908LF (will add it to the BOM aswell)

BernhardEllmann, 2020-06-14:
thanks to your here too !