LibreSolar/mppt-2420-rc #10: HV / LV side to 3.3V

BernhardEllmann, 2020-06-14:
The diagram on Page 3 page of PDF "mppt-2420-rc" (sheet "/Power Supply/") defines "HV +" as power input, which is linked to DCDC_HV+ on Page 1 (sheet "/"). 
Accordingly, the circuit board (version v0.1) is hardwired to DCDC_HV+

At the same time, Page 3 states: "Use voltage from low side if available."
This seems to offer a choice which is actually not available.

Please explain.

martinjaeger, 2020-06-14:
The Power Supply sheet uses the name HV+ internally, but it's connected to the DCDC_HV+ net on the top level schematic.

The circuit around Q9 in the Power Supply sheet "chooses" automatically, which rail (HV+ or LV+) it uses for internal power supply. If LV+ is available (>around 12V), it is used instead of HV+ to increase efficiency.

BernhardEllmann, 2020-06-14:
well, that's a nice feature :)