LibreSolar/mppt-2420-rc #11: boost operation mode?

BernhardEllmann, 2020-06-14:
I'm still wondering, if / why the controller can operate in boost mode (solar panel @18 V; Battery @ 36)

If I understood you correctly on the phone, this would involve connecting the battery to the HV side, and the solar panel to the LV side. But then the current would have opposite direction, right?
- Can the MOSFETS cope with that?
- What about "MOSFET for reverse polarity protection and PV reverse current blocking" (Schematic, Page 2, sheet "DC/DC", blue text next to Q8)

Is there any tutorial available?

martinjaeger, 2020-06-14:
Yep, see here: https://learn.libre.solar/development/dcdc_converter.html

The reverse polarity protection MOSFET will be switched on as soon as the DC/DC power stage is switched on. And if you use a connector where you can't connect the converter in reverse polarity, you can even use the terminal that goes directly to GNDPWR (marked W- for wind negative terminal).

BernhardEllmann, 2020-06-14:
nice.
But it does mean that the MOSFETs have to be able to cope with bidirectional current, right?
What about the in-built diodes in the MOSFETs, as indicated in the wiring symbols?

martinjaeger, 2020-06-14:
A MOSFET that is switched on does not care about the direction of current flow. And when it's off, current can only go in one direction (through the [body diode](https://en.wikipedia.org/wiki/Power_MOSFET#Body_diode)). And that's the same direction as you would have in a non-synchronous buck or boost converter.

BernhardEllmann, 2020-06-14:
ok, I think I got it now, thank you!

closed 2020-06-25 by martinjaeger